JoaoLopesF/SerialDebug #3: SerialDebug.cpp Missing include

wd5gnr, 2018-10-26:
SerialDebug.cpp lacks stdarg.h header which makes it fail to compile.

JoaoLopesF, 2018-10-26:
Hi @wd5gnr, any questions:
- The S.O. is Linux distro ?
- Arduino IDE or another IDE ?
- Arduino board

wd5gnr, 2018-10-26:
Compile with Arduino 1.8.5 for ARM board. I am under Linux but that shouldn't matter.

If you previously included varargs.h that has been deprecated and you must use stdarg.h now. Didn't check if that was the problem but adding stdarg.h under arduino.h fixes it.

JoaoLopesF, 2018-10-26:
Yes, thanks for response.
I just testing in Linux Mint and Arduino Mega, I do any tests with Due (ARM too)

wd5gnr, 2018-10-26:
Yes I have it working on an Uno and a Vidor MK4000 with the two changes (Arduino.h in 3 files and stdarg.h in this file.

JoaoLopesF, 2018-10-26:
Linux Mint, Arduino 1.8.7 and Due,  compile ok, not need that include.
Maybe it is due Arduino 1.8.5.
I  will put it in SerialDebug.cpp.
Please try it, before I relase new version:

/////// Includes

#include <Arduino.h>

#ifndef _STDARG_H // To avoid error in Arduino IDE 1.8.5 and linux - thanks to @wd5gnr 
	#include <stdarg.h>
#endif

JoaoLopesF, 2018-10-26:
Good, I just finish the update, please close it if always ok

closed 2018-10-28 by JoaoLopesF