JoaoLopesF/SerialDebug #4: "#error" compile error

BCChad, 2018-10-28:
**Describe the bug**
A clear and concise description of what the bug is.
Your two advanced examples have a statement "#error ..."  (see line 52) in a "#ifdef nest, that does not compile.  I can't seem to work around this.  If I comment out the statement, other compile errors are generated.  Please explain?

**To Reproduce**
Steps to reproduce the behavior:
1.    Please see line 52 of the advanced example for AVR
2.
3.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Arduino Information:**
 - OS: [e.g. Windows, MacOSX, Linux]
 - IDE [e.g. Arduino IDE, Eclipse, VSCode]
 - IDE Version [e.g. 1.8.6]
 - Board [e.g ESP32]

**Additional context**
Add any other context about the problem here.

per1234, 2018-10-28:
The board you are compiling is considered by the library to have low memory. Which board that is, I don't know because you didn't fill out the issue report correctly. When compiling for a "low memory" board, the library automatically goes into minimum mode, which the advanced example for AVR is not compatible with. If you want to use the library in non-minimum mode then you'll need to edit the file src/SerialDebug.h and comment out line 64:
https://github.com/JoaoLopesF/SerialDebug/blob/master/src/SerialDebug.h#L60-L66

BCChad, 2018-10-28:
Thanks for the explanation and work around, it helps me get acquainted with
the rich features of your debugger.
 For what it's worth my board is the Arduino Uno. I did try the  "Other "
example file and it gave a similar compile error.
Thanks very much .

JoaoLopesF, 2018-10-28:
Hi, @BCChad and @per1234 
@per1234, is correct your response, congrats.
@BCChad, please not use "Others" for Arduino Uno,
due this not use F() to reduce memory.
I will put more info in this #error. 
I will close this issue.
thanks a lot.

closed 2018-10-28 by JoaoLopesF