JoaoLopesF/SerialDebug #5: strings corrections

lpdunwell, 2018-11-08:
hi there, i like this project!
i'd like to offer doing some proofreading, there's a couple of typos (especially in the app).
if you're interested, tell me!

harshitc, 2018-11-16:
I found this app today and it seems to be a good one. Haven't tried it yet but I found a typo in the app(version 0.9.4) under the settings tab. The apply is spelled as "aplly". I will surely give this app a try whenever I get the time to be on Arduino IDE.

Thanks
Hc

lpdunwell, 2018-11-16:
the thing is, i can't seem to find the source to the app itself, and digging through what i can download yields no real starting point. so if you tell me where to begin, i'd do this in like a day tops.
cheers, luke

lpdunwell, 2018-11-16:
if you want to leave it closed, send me a complete list of strings, i'll return them corrected.

lpdunwell, 2018-11-19:
got it, working on it!
cheers

lpdunwell, 2018-11-19:
ok i'm not entirely sure about a few of them, here's the lot as of now.
i'll go over it again, trying to figure out the ones i'm uncertain about and probably catch what i missed.
i copied what you sent me, replaced ";" with ";\n" and removed leading spaces before doing anything.
i hope this doesn't make things too messy on your side. if anything i did doesn't make sense whatsoever, just tell me :D
cheers

//snip

serialDebugAppUI.exibirPopupAlerta("<html>Detected possible serial baud error,<br>please disconnect and correct this.</html>", 10);
ret += " " + ((enabled)?"(enabled)":"(disabled)") + ((triggered)?" (triggered)":"");
serialDebugAppUI.exibirPopupAlerta("Text sent to clipboard.", 3);
serialDebugAppUI.exibirPopupAlerta("File saved at: " + arquivo, 5);
fcArquivo.setDialogTitle("Save text file");
serialDebugAppUI.btnFiltrar.setText("Cancel");
serialDebugAppUI.btnFiltrar.setText("Filter");
serialDebugAppUI.exibirPopupAlerta("No serial ports detected", 3);
serialDebugAppUI.exibirPopupAlerta("No serial speed (baudrate) selected", 3);
serialDebugAppUI.exibirPopupAlerta("Invalid serial speed (baudrate) - non numeric", 5);
serialDebugAppUI.exibirPopupAlerta("Invalid serial speed (baudrate) - not positive numeric", 5);
adicionarNaLista("* *** Connected to serial port USB: " + ultPortaSerial, true);
adicionarNaLista("* Discarding buffer from serial", true);
adicionarNaLista("* Done", true);
serialDebugAppUI.exibirPopupAlerta("Not possible to connect to " + serialPort.getPortName(), 5);
adicionarNaLista("* *** Disconnected from serial port USB", true);
serialDebugAppUI.exibirPopupAlerta("The serial port USB not is connected", 5);
serialDebugAppUI.exibirPopupAlerta("Not possible to send data to serial port", 5);
serialDebugAppUI.exibirPopupAlerta("Exception on send data to serial port", 5);
serialDebugAppUI.btnNonStop.setText("W.nonstop");
serialDebugAppUI.btnNonStop.setToolTipText("To watch non stop");
serialDebugAppUI.btnNonStop.setText("W.stop");
serialDebugAppUI.btnNonStop.setToolTipText("To watch stop");
serialDebugAppUI.popUpUpdate.lblUpdateTitulo.setText("Checking for updates ...");
serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugApp.setText(" App.: checking ...");
serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugLib.setText(" Lib.: checking ...");
serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugApp.setText(" App.: " + versaoAppInternet + "(latest) " + SerialDebugApp.VERSAO_MAIOR + "." + SerialDebugApp.VERSAO_MENOR + "." + "(installed) ");
serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugApp.setText(serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugApp.getText() + "(needs update)");
serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugApp.setText(serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugApp.getText() + "(ok)");
serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugLib.setText(" Lib.: " + versaoLibInternet + "(latest) ");
serialDebugAppUI.lblInfoLibVersao.getText() + "(on board) ");
serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugLib.setText(serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugLib.getText() + "(needs update)");
serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugLib.setText(serialDebugAppUI.popUpUpdate.lblUpdateSerialDebugLib.getText() + "(ok)");
serialDebugAppUI.popUpUpdate.esconder("Versions checked", 7);
serialDebugAppUI.popUpUpdate.lblUpdateTitulo.setText("Versions checked");
serialDebugAppUI.popUpUpdate.esconder("Connection not possible", 5);
static String[] colunas = new String[]{"Number", "Name", "Type", "Value", "Description"};
static String[] colunas = new String[] { "Number", "Name", "Arg.type", "Last arg.", "Description" };
public final String titulo = "End User License Agreement (EULA)";
btnAceita = new JButton("Accept");
JButton btnNotAccept = new JButton("Do not accept");
textArea.setToolTipText("SERIALDEBUGAPP FREEWARE LICENSE AGREEMENT");
"SERIALDEBUGAPP FREEWARE LICENSE AGREEMENT\n--------------------------------------\nExcept otherwise noted, all of the documentation and software included\nin the SerialDebugApp distribution is copyrighted \nby Joao Clovis Lopes Filho, developer known as JoaoLopesF.\nShould you have any questions regarding this license or the software included,\nplease email contact@joaolopesf.net\n\n\nImportant - Read Carefully\n--------------------------\nThis license agreement is a legal agreement between you (either an\nindividual or a single entity) and JoaoLopesF, regarding the use of\nthe SerialDebugApp distribution. By installing, copying, or otherwise using\nthe software, you agree to be bound by the terms of this agreement. If\nyou do not agree to the terms of this agreement, do not install or use\nthe software.\n\n\nFREEWARE LICENSE\n----------------\nThis software is being distributed as Freeware. It may be freely used,\ncopied and distributed as long as it is not sold, and all original\nfiles are included in their original form, including this license. You\nare NOT allowed to make a charge for distributing this software\n(either for profit or merely to recover your media and distribution\ncosts) whether as a stand-alone product, or as part of a compilation\nor anthology, without explicit prior written permission. By using this\nsoftware, you agree to these terms and the terms of the Disclaimer\nbelow:\n\nDisclaimer of Warranty\n---------------------------------------------------------------- \nTHIS SOFTWARE IS PROVIDED AS IS AND WITHOUT WARRANTY OF ANY KIND\nEXPRESSED, IMPLIED OR OTHERWISE, INCLUDING WITHOUT LIMITATION, ANY\nWARRANTY OF MER [... cut ...]
printWriter.println(" // SerialDebug handle");
printWriter.println(" // Note: if in inactive mode (until anything received from serial),");
printWriter.println(" // it shows only messages of 'always' or 'error' levels type");
printWriter.println(" // and the overhead during inactive mode is very low");
printWriter.println(" // Only if not DEBUG_DISABLED");
printWriter.println("");
printWriter.println(" debugHandle();
");
printWriter.println("");
lblConversion.setText("File converted successfully.");