JoaoLopesF/SerialDebug #7: Moving towards subset of GDB as a serial-line protocol?

phrxmd, 2019-01-23:
The library is great and does a good job of providing logging and a basic debugger over the serial line. I would love to see this become more full-featured in the future - some other common debugging features might be easy to implement, such as breakpoints.

That said, I think it would be convenient if the command protocol for debugging was following a subset of the GDB protocol (maybe optional using a `#define`). The GDB protocol is super-simple, based around single-letter commands much similar to what you are already doing now. That way one could reuse the various host development tools made around `gdb`. Of course it would never be the whole range of everything that `gdb` does, but even a subset would be useful.

JoaoLopesF, 2019-01-23:
Hi phrxmd,
I am glad that you like this.
A question: I am use GDB with ESP32, but I always need extra USB hardware, compatible with openocd,
how us do it, if the purpose of this libray, is debug wireless ?

I made one library, to improve debugging in Arduino, without extra hardware:

https://github.com/JoaoLopesF/SerialDebug

In future, I add this resources to RemoteDebug.

Your help is very good to improve more the RemoteDebug.

Thanks a lot

phrxmd, 2019-01-23:
Hi @JoaoLopesF, and thanks again for your good work!
Yeah, I had SerialDebug in mind, that's why I'm commenting here :) RemoteDebug is a different story altogether. 

The GDB protocol is just a simple protocol with a basic extensible command set. It's what debugging software uses to interface with OpenOCD. Having GDB access to SerialDebug over the serial link wouldn't give all the low-level access you get with JTAG hardware. But it would allow reusing your existing debugging tools for the basic debugging functions in SerialDebug gets you.

JoaoLopesF, 2019-01-24:
Hi @phrxmd, I will research on it, but I see a problem: map of memory for objects, in ESP32 it is done reading the elf binary file, but how do it without openocd ?
And I will implement the debugger of SerialDebug on RemoteDebug. 
Tks

phrxmd, 2019-01-24:
Hi @JoaoLopesF, I think that's why it's going to be a subset, making available only what is accessible from the program logic itself, but using the same command interface - the GDB protocol is just a set of basic commands in plain text over a serial line and not all of them have to be supported right away.

phrxmd, 2019-02-08:
@JoaoLopesF sorry for the late reply, it seems like I missed your answer here.

Here are a few samples and tutorials of remote GDB use from various contexts:
- Stackexchange: [How can I run gdb in one machine and kgdb kernel in another?](https://unix.stackexchange.com/questions/105222/how-can-i-run-gdb-in-one-machine-and-run-kgdb-kernel-in-another-machine), GDB protocol over serial port
- NetBSD Documentation: [Debugging the NetBSD kernel with GDB HOWTO](https://www.netbsd.org/docs/kernel/kgdb.html), GDB protocol over serial port
- Tutorial: [Debugging Linux kernel using gdb under VirtualBox](http://sysprogs.com/VBoxGDB/tutorial/), GDB protocol over TCP/IP

There is also a bit of documentation in implementing the server side of the GDB remote server protocol in various OSes: 
- [Howto: GDB Remote Serial Protocol / Writing a RSP Server](https://www.embecosm.com/appnotes/ean4/embecosm-howto-rsp-server-ean4-issue-2.html) (general tutorial)
- Kitten OS: [Implementing a GDB Stub in Lightweight Kitten OS](http://people.cs.pitt.edu/~anz28/papers/2013_kitten_gdb_tr.pdf) (PDF on implementing a GDB server in an existing embedded OS)