JoaoLopesF/SerialDebug #8: Support Serial1 port on ESP8266

sgrimee, 2019-04-11:
**Describe the solution you'd like**
When running a serial application on ESP8266 the main port Serial cannot be used for debugging. An alternative is to use Serial1 on another pin as a write-only UART to log messages. It would be great to be able to use SerialDebug on this UART

**Describe alternatives you've considered**
Your RemoteDebug library to log over TCP is awesome. Sometimes however, Wifi is not an option.

**Additional context**
Ideally, being able to use *both* SerialDebug on UART1 and RemoteDebug when wifi is available would be awesome.

JoaoLopesF, 2019-05-09:
Hi, sorry for this large daly on response.

Serial1 of ESP8266 cannot be used, due it have only tx, and some features use rx to work.

You can use SoftwareSerial to comunicate and hardware serial to debug.

Regards

sgrimee, 2019-05-10:
thanks Joao, understood. I was thinking about the logging features only, without debugger. For this making the serial port configurable would still be useful...

afefer, 2019-12-31:
Hi Joao - I have a similar issue with the Teensy 3.5. Serial() defaults to the USB serial port, which isn't available when I'm using the USB port in a different mode. As a result I want to use Serial1 (which is a full-fledged UART with rx/tx capabilities). Is there any way to select a different serial port when configuring SerialDebug?

arkhipenko, 2020-07-07:
Chiming in here. 
ESP32 has 3 UARTS, and I am currently working on a project that uses Serial and Serial1 for functionality, leaving Serial2 for debug. 
It would be very useful to be able to define which UART you want the output on! Even for Tx only.

jpmeijers, 2020-11-03:
This is also getting my vote.

Some boards like the Arduino Zero has a Serial and SerialUSB. If I want to use SerialUSB I have to modify the board definitions so that Serial alias SerialUSB. That's less than ideal. It would be great if this library allowed passing in the Serial object or Stream that should be used.

nerdyscout, 2024-03-04:
I made a draft for this request and hope a few people could try this on different boards:
https://github.com/nerdyscout/SerialDebug/tree/feature/customSerial